DavidVentura/cam-reverse #2: Package dependiences

mickeyreg, 2024-04-04:
On the first try I had:
```
$ make run
npm install
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'mocha@10.2.0',
npm WARN EBADENGINE   required: { node: '>= 14.0.0' },
npm WARN EBADENGINE   current: { node: 'v12.22.9', npm: '8.5.1' }
npm WARN EBADENGINE }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'typescript@5.3.3',
npm WARN EBADENGINE   required: { node: '>=14.17' },
npm WARN EBADENGINE   current: { node: 'v12.22.9', npm: '8.5.1' }
npm WARN EBADENGINE }

added 97 packages, and audited 98 packages in 5s

20 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
./node_modules/.bin/ts-node --esm http_server.ts
/home/marcinkk/cam-reverse/node_modules/typescript/lib/typescript.js:139
    for (let i = startIndex ?? 0; i < array.length; i++) {
                             ^

SyntaxError: Unexpected token '?'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at attemptRequireWithV8CompileCache (/home/marcinkk/cam-reverse/node_modules/ts-node/dist/util.js:104:16)
    at loadCompiler (/home/marcinkk/cam-reverse/node_modules/ts-node/dist/configuration.js:237:56)
    at resolveAndLoadCompiler (/home/marcinkk/cam-reverse/node_modules/ts-node/dist/configuration.js:226:16)
make: *** [Makefile:16: run] Error 1
```
I thought it is the version problem, so after some searching I've made:
```
npm install typescript@5.0.4
npm install mocha@9.2.2
```
and the server started.

Maybe proper/required versions of the packages can be implemented somewhere in the code?

mickeyreg, 2024-04-04:
Hmmm... I'm newbie in npm and node :)
```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.4 LTS"
```
I have `nodejs v12.22.9` installed globally with `apt`. This version was in the error displayed after first type of `make run`.

I can install node locally `npm install node@18.20.1`, but it does not help.

```
$ npm ls
cam-reverse@ /.../cam-reverse
├── mocha@9.2.2
├── ts-node@10.9.2
├── node@18.20.1
└── typescript@5.0.4
```

mickeyreg, 2024-04-06:
I've installed additional repository in the Ubuntu and then I installed nodejs v18.20.1 using apt - cam-reverse server won't start.

It starts fine on v18.16.0.

The above was for testing. On the target system, Ubuntu Server 22.04 LTS, the app works fine with nodejs v12.22.9 - default on this system.

adrcunha, 2024-04-24:
For the record, node v14.15.5 and npm v6.14.11 on macOS Monterey (12.6.6) worked fine, although I had to run `npm i @types/node` before successfully running the HTTP server.

DavidVentura, 2024-04-26:
I've added a build / bundle step in CI in this repo, you should be able to download the bundles from here:
https://github.com/DavidVentura/cam-reverse/actions/runs/8845445916
and run with `node bundle.cjs` -- I've tried with Node12 & Node18 on Linux and it seems to work

closed 2024-04-26 by DavidVentura