DavidVentura/cam-reverse #3: Few comments on documentation :)

mickeyreg, 2024-04-06:
I started using your server and it's workind fine. I can connect to the camer without YsxLite. I have 2 X5. both works. But I have to take a look into sources :)

https://github.com/DavidVentura/cam-reverse/blob/cf45216693734e58f50cc36f73e43beedd65cbf2/http_server.ts#L14

`discovery_ip` has to be changed to get camera visible - broadcast addres of the user network or IP of the camera

I coulndn't pair the camera with my WiFi. I connected with the AP created by the camera and the pairing wasn't successful. I paired both my cameras with YsxLite and then blocked them from the Internet on firewall ;)

The first problem i that the structure `opts`: 
https://github.com/DavidVentura/cam-reverse/blob/cf45216693734e58f50cc36f73e43beedd65cbf2/cmd/pair.ts#L7
is lack from the element `attempt_to_fix_packet_loss`.
I added it and then the discovery prosess started, but without success :(

DavidVentura, 2024-04-08:
What do you mean with "pairing wasn't successful"? did you get some logs?
The only thing I can imagine is that there's a hardcoded `/24` for the network mask here:
https://github.com/DavidVentura/cam-reverse/blob/cf45216693734e58f50cc36f73e43beedd65cbf2/impl.ts#L128

mickeyreg, 2024-04-09:
If I understood correctly:

After resetting camera to "factory default" I should conned with the AP created by camera and then run:
```
SSID=<your ssid> PSK=<your password> make pair
```
and as a result the camera should be configured with my wifi network parameters and then restarted. I do not need to use YsxLite for configuration?

About logs: I can send what is generated on the screen. Is there something else then cosole.log?

DavidVentura, 2024-04-14:
> . I do not need to use YsxLite for configuration?
Indeed, I've only used YsxLite once to reverse the packet contents, but not for the other cameras.

Is there anything useful on the logs? Do you see the camera restart after the pairing attempt?

adrcunha, 2024-04-24:
I faced the two issues mentioned in the first post, https://github.com/DavidVentura/cam-reverse/pull/7 and https://github.com/DavidVentura/cam-reverse/pull/6 solved those for me.

But I did have to run `make pair` a few times until it finally succeeded (several places mention that the camera isn't very reliable, even when using the recommended apps). I wasn't able to configure it for a no-password network though (and didn't investigate further once it worked with SSID/password).

DavidVentura, 2024-04-24:
Yeah, I don't expect the current implementation to necessarily work without a password - there's a dedicated part of the payload to it. It may work with the camera's firmware if the value is zero-length, but I've not sniffed that from the app

adrcunha, 2024-04-25:
I played with the ftycampro app and couldn't make it connect to a network without a password. It seems that the app supports it, but not the camera (at least not the model I have): after it reboots it never connects to the network; everything works as expected if I set the network to one protected by a password.

DavidVentura, 2024-04-26:
In this commit: 639a78419eaef6f0afdc623feffad3b5fc5dddeb

I've added the missing options, added a flag to allow specifying the discovery ip address and blocked using an empty password

closed 2024-04-26 by DavidVentura