DavidVentura/cam-reverse #4: Some questions :)

mickeyreg, 2024-04-06:
Some questions and comments on the X5 camera itself, partially connected with the cam-reverse app.

Generally X5 camera is rather useless ;) Even with the YsxLite works in the randomly - sometiems connects, sometimes not :)

OK. I configured and run cam-reverse. It discovered both my cameras and I added them to the HomeAssistant (links to nodejs server). Works, but after some time one or another camera dissaperars. On the server screen it looks the same:
```
requested for BATC631408JMENX
requested for BATC631486CDCID
Conn closed, kicked
Conn closed, kicked
.
.
.
.
.
```
Restart of the server does not help. Power cycle on the camera itself is required to get it back.

Is it possible to repair this issue in the cam-reverse server code?

The cam-reverse stremas MJPEG with 640x480 resolution. Is the higher resolution available? Is it possible to get rtsp stream directly from the camera?

DavidVentura, 2024-04-08:
I've not have had the issue with the X5 camera, at least on my overnight tests.
The resolution is only 640x480 or 320x240 sadly, you can check the reversed implementation here: https://github.com/DavidVentura/cam-reverse/blob/master/impl.ts#L87

but it seems like the cameras do not really respect the protocol

mickeyreg, 2024-04-09:
I had the issue, but last 2 days both cameras working without problems :) So it looks like random and unpredictable problem :(

BTW: I created systemd scripts for cam-reverse. They base on James Chambers' work: https://github.com/TheRemote/MinecraftBedrockServer

So, step by step :)

First I created user camera in my Ubuntu Server and download cam-reverse from GitHub to user directory.

Then I created systemd script (as a root): `nano /etc/systemd/system/cam-reverse.service`
```
[Unit]
Description=cam-reverse server
After=network-online.target

[Service]
User=camera
WorkingDirectory=/home/camera/cam-reverse
Type=forking
ExecStart=/bin/bash /home/camera/start.sh
ExecStop=/bin/bash /home/camera/stop.sh
GuessMainPID=no
TimeoutStartSec=600

[Install]
WantedBy=multi-user.target
```
and two bash scripts (as the camera user):
`nano /home/camera/start.sh`
```
#!/bin/bash

# Set path variable
USERPATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
PATH="$USERPATH"

# Change directory to server directory
cd /home/camera/cam-reverse

# Check if network interfaces are up
NetworkChecks=0
DefaultRoute=$(/sbin/route -n | awk '$4 == "UG" {print $2}')
while [ -z "$DefaultRoute" ]; do
    echo "Network interface not up, will try again in 1 second";
    sleep 1;
    DefaultRoute=$(/sbin/route -n | awk '$4 == "UG" {print $2}')
    NetworkChecks=$((NetworkChecks+1))
    if [ $NetworkChecks -gt 20 ]; then
        echo "Waiting for network interface to come up timed out - starting server without network connection ..."
        break
    fi
done

echo "Starting cam-reverse server.  To view window type screen -r cam-reverse"
echo "To minimize the window and let the server run in the background, press Ctrl+A then Ctrl+D"

BASH_CMD="./node_modules/.bin/ts-node --esm http_server.ts"

# screen -L -Logfile logs/cam-reverse.$(date +%Y.%m.%d.%H.%M.%S).log -dmS cam-reverse /bin/bash -c "${BASH_CMD}"
screen -dmS cam-reverse /bin/bash -c "${BASH_CMD}"
```
`nano /home/camera/stop.sh`
```
#!/bin/bash

# Set path variable
USERPATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
PathLength=${#USERPATH}

# Check if server is running
if ! screen -list | grep -q "\.cam-reverse"; then
  echo "Server is not currently running!"
  exit 1
fi

# Stop the server
echo "Stopping cam-reverse server ..."

if screen -list | grep -q "\.cam-reverse"; then
  screen -S cam-reverse -X quit
fi

echo "cam-reverse server stopped."
```

And finally the scripts was activated:
```
chmod 755 /home/camera/start.sh
chmod 755 /home/camera/stop.sh
systemctl daemon-reload
systemctl enable cam-reverse
``` 
and started:
```
systemctl start cam-reverse
```
if required:
```
systemctl status cam-reverse
systemctl stop cam-reverse
```
and maybe:
```
sudo su camera
screen -r cam-reverse
```
Scripts requires net-tools: `apt install net-tools`

closed 2024-04-26 by DavidVentura