DavidVentura/cam-reverse #5: Unknown file extension ".ts"

Zootsuit7, 2024-04-21:
What am I doing wrong?
`jim@jim-lenovo:~/cam-reverse-master$ make run
./node_modules/.bin/ts-node --esm http_server.ts
TypeError [ERR_UNKNOWN_FILE_EXTENSION]: Unknown file extension ".ts" for /home/jim/cam-reverse-master/http_server.ts
    at Object.getFileProtocolModuleFormat [as file:] (node:internal/modules/esm/get_format:160:9)
    at defaultGetFormat (node:internal/modules/esm/get_format:203:36)
    at defaultLoad (node:internal/modules/esm/load:143:22)
    at async nextLoad (node:internal/modules/esm/hooks:866:22)
    at async nextLoad (node:internal/modules/esm/hooks:866:22)
    at async Hooks.load (node:internal/modules/esm/hooks:449:20)
    at async MessagePort.handleMessage (node:internal/modules/esm/worker:196:18) {
  code: 'ERR_UNKNOWN_FILE_EXTENSION'
}
make: *** [Makefile:16: run] Error 1`
`jim@jim-lenovo:~/cam-reverse-master$ node -v`
`v20.12.2`
`jim@jim-lenovo:~/cam-reverse-master$`

Zootsuit7, 2024-04-21:
So I installed tsx-
`npm i -D tsx`
Then I changed the line in Makefile accordingly-
Originally-
`run: node_modules
	./node_modules/.bin/ts-node --esm http_server.ts`
Now-
`run: node_modules
	npx tsx http_server.ts`
Save changes and-
`make run`
Hope this helps somebody.

DavidVentura, 2024-04-26:
I've added a build / bundle step in CI in this repo, you should be able to download the bundles from here:
https://github.com/DavidVentura/cam-reverse/actions/runs/8845445916
and run with `node bundle.cjs` -- I've tried with Node12 & Node18 on Linux and it seems to work

closed 2024-04-26 by DavidVentura