DavidVentura/cam-reverse #14: Clarification about the Spyware section

adrcunha, 2024-04-27:
### Background/extra info

The "Spyware" section in the README mentions 4 IPs where a HELLO message is sent to, and they're stored inside the YsxLite app APK. The README also mentions other IPs, some already extracted in the `dec_svr.py` file. Right now, they map to the following:

SWPNPDPFLVAOLNSXPHSQPIEOPAIDENLXHXEHIFLKPGLRHUARSTLQEEEPSUIHPDLSPEAOICLOSQEMLPPALNIBIAERHZLKHXEJHYHUEIEHELEEEKEG
* 139.155.68.77: not found: 2(SERVFAIL)
* 119.45.114.92: not found: 2(SERVFAIL)
* 162.62.63.154: not found: 2(SERVFAIL)
* 3.132.215.40: ec2-3-132-215-40.us-east-2.compute.amazonaws.com.

PFLXLSTBLKHYLPLRHUEHIEEGEEARLQPLIHIAEKAOSTLVEOSQPDHZLNPAICIFEREJLKEMENHUHXIBEPEEEHEIEL
* 69.235.180.149: ec2-69-235-180-149.cn-northwest-1.compute.amazonaws.com.cn.
* 44.198.131.69: ec2-44-198-131-69.compute-1.amazonaws.com.
* 3.67.159.125: ec2-3-67-159-125.eu-central-1.compute.amazonaws.com.

EKTDROREHXHURHRKRMCXEEKPRNKKUZPNLXNYNOHYAONRNUNWRJSQGZNXGUNTIHKIJYEHPAKBHTKEKGKDLKJVKHKFERGSGIEIHUGLEDGOGQGNEEGFGRGP
* pf1.ilnk.work: private registration through GoDaddy
* pf2.ilnk.work: private registration through GoDaddy
* pf1.i-lnk.com: private registration through Alibaba
* pf2.i-lnk.com: private registration through Alibaba

EKPNHXIDAUAOEHLOTBSQEJSWPAARTAPKLXPGENLKLUPLHUATSVEESTPFHWIHPDIEHYAOLVEISQLNEGLPPALQHXERELIALKEHEOHZHUEKIFEEEPEJ
* 174.129.34.188: ec2-174-129-34-188.compute-1.amazonaws.com.
* 47.89.33.130: not found: 2(SERVFAIL)
* 182.92.103.41: not found: 2(SERVFAIL)
* 54.183.49.93: ec2-54-183-49-93.us-west-1.compute.amazonaws.com.

PFLXLNPKSULKLVPEHUHXHWLPEEPGEHIHARENLQAOSTLOIFSQHZIAPAPDLUIEERLNEOHYLKEPEIHUHXEGEJEEEKEH
* 182.92.103.41: not found: 2(SERVFAIL)
* 174.129.34.188: ec2-174-129-34-188.compute-1.amazonaws.com.
* 182.92.103.41: not found: 2(SERVFAIL)

SVLXLNENPGLKHXLOLVHUPFLQEEEHPKLUIHIALPAOARHYIESQSTEKIBPAPDEIHZERLNEJIFLKHXEOEMHUEHELHWEEEPEG
* 69.235.180.149: ec2-69-235-180-149.cn-northwest-1.compute.amazonaws.com.cn.
* 44.198.131.69: ec2-44-198-131-69.compute-1.amazonaws.com.
* 3.67.159.125: ec2-3-67-159-125.eu-central-1.compute.amazonaws.com.

I found a subset of the exact strings in the FtyCampro app:

EKPNHXIDAUAOEHLOTBSQEJSWPAARTAPKLXPGENLKLUPLHUATSVEESTPFHWIHPDIEHYAOLVEISQLNEGLPPALQHXERELIALKEHEOHZHUEKIFEEEPEJ
* 174.129.34.188: ec2-174-129-34-188.compute-1.amazonaws.com.
* 47.89.33.130: not found: 2(SERVFAIL)
* 182.92.103.41: not found: 2(SERVFAIL)
* 54.183.49.93: ec2-54-183-49-93.us-west-1.compute.amazonaws.com.

PFLXLNPKSULKLVPEHUHXHWLPEEPGEHIHARENLQAOSTLOIFSQHZIAPAPDLUIEERLNEOHYLKEPEIHUHXEGEJEEEKEH
* 182.92.103.41: not found: 2(SERVFAIL)
* 174.129.34.188: ec2-174-129-34-188.compute-1.amazonaws.com.
* 182.92.103.41: not found: 2(SERVFAIL)

SVLXLNENPGLKHXLOLVHUPFLQEEEHPKLUIHIALPAOARHYIESQSTEKIBPAPDEIHZERLNEJIFLKHXEOEMHUEHELHWEEEPEG
* 69.235.180.149: ec2-69-235-180-149.cn-northwest-1.compute.amazonaws.com.cn.
* 44.198.131.69: ec2-44-198-131-69.compute-1.amazonaws.com.
* 3.67.159.125: ec2-3-67-159-125.eu-central-1.compute.amazonaws.com.

SVLXLNENPGLKHXLOLVHUPFLQEEEHPKLUIHIALPAOARHYIESQSTEKIBPAPDEIHZERLNEJIFLKHXEOEMHUEHELHWEEEPEG
* 174.129.34.188: ec2-174-129-34-188.compute-1.amazonaws.com.
* 43.128.145.123: not found: 2(SERVFAIL)
* 139.186.150.90: not found: 2(SERVFAIL)

SWPNPDPFLVAOLNSXPHSQPIEOPAIDENLXHXEHIFLKPGLRHUARSTLQEEEPSUIHPDLSPEAOICLOSQEMLPPALNIBIAERHZLKHXEJHYHUEIEHELEEEKEG
* 139.155.68.77: not found: 2(SERVFAIL)
* 119.45.114.92: not found: 2(SERVFAIL)
* 162.62.63.154: not found: 2(SERVFAIL)
* 3.132.215.40: ec2-3-132-215-40.us-east-2.compute.amazonaws.com.

These generic catch-all camera apps seem to be all derived from the same source, changing just the UI and some bits of behavior.

### Ping origin: camera or app?

I don't have a setup to trace the attempts to send the HELLO messages to these IPs, but they're found in the APKs and the README states that "connecting the camera to a network, it tries to send a HELLO". I'd like to confirm the source of the messages, so the documentation can be updated about optionally blocking these IPs on the router (but hopefully they're sent by the app and thus using cam-reverse is totally safe).

@DavidVentura were you able to confirm the source of the messages, if they are sent by the app or by the camera firmware?

DavidVentura, 2024-04-27:
These strings can be uploaded to the camera; and indeed, my camera was the one generating the requests.
This is the 'feature' used to be able to see your live streams over the internet. I've not investigated how to overwrite these values, instead I block all outgoing traffic from the cameras

adrcunha, 2024-04-27:
Thanks for the details. Sent PR #15 to add these details to the README.

closed 2024-04-27 by adrcunha