DavidVentura/cam-reverse #22: New option to skip a camera

anddag, 2024-05-05:
In my network there are 3 cameras that use the ILINKP2P protocol. One of them is very old (detected but the video dont is shown) and I would like to exclude it from the search. Could you implement an option  to esclude some IP o Cameras from discovery and handshake?

DavidVentura, 2024-05-05:
The serial number is only known after handshake, so it'd be hard to ignore devices before then. Ignoring based on IP address would be easier

anddag, 2024-05-05:
Ok for ip Based :-)

DavidVentura, 2024-05-05:
It's now pushed, with d5d4161dd8892cdecd158201f65615685065f5a5

You will need to provide a settings file and `--config_file` for the webserver.

You can provide jsut the blacklist key, eg :
```yaml
blacklisted_ips:
  - 192.168.1.81
```

check [the config sample file](https://github.com/DavidVentura/cam-reverse/blob/master/config.example.yml) for all settings

closed 2024-05-05 by DavidVentura