DavidVentura/cam-reverse #28: Dont find the device...

danielandradam, 2024-06-06:
my gateway network is 192.168.0.1 in this case do I have to modify something?

I didn't understand the help well. Would I have to find out the IP address of the camera?? But how to do this before it pairs with the app??

Help me please

Message from bin.cjs...

Searching for devices on 192.168.1.255

DavidVentura, 2024-06-06:
Hey Daniel, you should set discovery_ips to 192.168.0.255 most likely:

```yaml
 discovery_ips:
   - 192.1680.255
```

danielandradam, 2024-06-06:
Thank you. I Will try! How long time is necessary to find a device? More one question..is necessary that the mini camera stay paired with app before turn on ap mode? Congratulations for the great project!

nikto-b, 2024-06-24:
in case if discovery does not helps, you can search for devices with an arp scanning ([such as](https://github.com/royhills/arp-scan)) after pairing, and pass an ip of your camera into discovery_ips directly  
As for me, it was necessary to edit `discovery_ips` in `settings.ts` (and rebuild project) for a pair command just to find device and pair it to my network

DavidVentura, 2024-06-28:
@nikto-b you can set discovery_ips in config.yml, no need to rebuild

@danielandradam are you able to capture traffic between the camera and the server to see where things go wrong?
have you tried connecting to the camera's AP and using the `http_server` command to see if that works at all? (to verify if it's a pair issue or streaming also does not work)