DavidVentura/cam-reverse #30: Can't pair with my camera

mtiahg, 2024-06-28:
Hello!
I'm trying to pair my TXW806-840 but I can't get any success. Maybe the code doesn't work with this model?
When connected to the pairing camera hotspot, my gateway is 192.168.1.1, so scanning for 192.168.1.255 would work, right?

I also tried to change the discovery ip to 192.168.1.1 but it continued not working.
I confirmed in wireshark that the camera is indeed 192.168.1.1.

Maybe you could help me?

Thanks

DavidVentura, 2024-06-28:
could you capture traffic between the camera and this server to see where it goes wrong?
ideally, could you also capture traffic between the camera and the official app, to compare the two?

acdcl, 2024-07-09:
Hi David! Thanks for the great work you have done! Congrats! I am afraid mtiahg is facing the same problem I am have with my cameras (DGOA..., DGOE..,,). The first one comes with Beken 7252 and the second with txw817-810. I capture some traffic using wireshark and realized that you app broadcast f1 30 00 00 00 to get a punch from the camera, but there was no reply. Using a9serv C app, I realized that it sends 2c ba 5f 5d, and in this case the communication with the camera is estabished. I try to changing the search message in our code, but didn't have success.

acdcl, 2024-07-09:
Hear are snapshots of wireshark.
[image]

acdcl, 2024-07-09:
[image]

acdcl, 2024-07-09:
Any thoughts? I am trying to help to include the new serie of A9 cams that are availabe in the market nowadays. Thanks for time!

DavidVentura, 2024-07-09:
can you send me a link to where you got the cameras? I'll get one and test it out.

If the discovery message is the only difference, you could test modifying this function:

https://github.com/DavidVentura/cam-reverse/blob/master/impl.ts#L176

to be
```ts
export const create_LanSearch = (): DataView => {
  const outbuf = new DataView(new Uint8Array(4).buffer);
  outbuf.writeU16(0x2c);
  outbuf.writeU16(0xba);
  outbuf.writeU16(0x5f);
  outbuf.writeU16(0x5d);
  return outbuf;
};
```

acdcl, 2024-07-09:
I change the code and got it using wireshark
[image]

acdcl, 2024-07-09:
Solved with:
export const create_LanSearch = (): DataView => {
  const outbuf = new DataView(new Uint8Array(4).buffer);
  outbuf.writeU32(0x2cba5f5d);
  return outbuf;
};

acdcl, 2024-07-09:
Camera punch:
[image]

acdcl, 2024-07-09:
That's it! App keeps sending 2cbabfbd and camera replying 2ccb6293bf2321ed0ad7ea318106e0d5a28d800233207369.

DavidVentura, 2024-07-10:
I've also gotten my camera from AliExpress, so it's a bit of a gamble whether buying that listing will land me with a camera like yours

I've just pushed a branch `encoded` (https://github.com/DavidVentura/cam-reverse/tree/encoded), can you try it out? I can't really test it, but it should apply the encoding/decoding as a9server does.

acdcl, 2024-07-10:
Sure! I’ll have a try, thanks.