DavidVentura/cam-reverse #31: Instructions unclear

Leblayd, 2024-06-28:
I seriously do not understand what I should be doing here.

I bought an A9 model. Tried it with the official app, and it works.
Now I wanna use it directly (for HomeAssistant, ultimately).

1. I clicked remove in the app, so it doesn't interfere.
2. I turned on the camera and entered AP mode.
3. I connected to that AP from my PC
4. I built this project

Everything is clear until now. But after that, I'm confused.

**HTTP Server** 
I ran `node dist/bin.cjs http_server`, and but nothing happens.  
Tried modifying the config file so that it checks other IPs for discovery.  
It did not work.
Apparently you have to pass the config explicitly with `--config_file config.example.yml` (please add this to the Readme)
I added `192.168.0.1`, because that's what I see in `ip route`, and `192.168.0.255` because it was posted in another issue here.
Now it's checking every IP address, and it is in debug mode.
However still, nothing happens whatsoever.
No error in the console, either.

**Pairing**
I struggled to understand what I should even pass as SSID and Password, until I realized you must mean my proper (2.4ghz) Wifi network.
I ran `node dist/bin.cjs http_server --ssid what --password ever`, and nothing happens.
Tried passing `--config_file config.example.yml` but didn't help.
Saw in the code that you're not actually loading the config from the command line parameters, so the config object is always going to be the default, which seems like a mistake. (if so, please fix)
Fixed it and rebuilt the project.
Now it's checking each IP, but still no change, or even a debug message.

Tried the same thing again after re-doing everything in the app so that it might already be "paired" with my Wifi, but the result is the same.

What am I even supposed to do?

DavidVentura, 2024-06-28:
>  I ran node dist/bin.cjs http_server, and but nothing happens.
what do you mean by nothing happens? do you get any output at all?
I get 
```
$ node dist/bin.cjs http_server
2024-06-28T21:29:10.536Z [info] Starting HTTP server on port 5000
2024-06-28T21:29:10.542Z [info] Searching for devices on 192.168.1.255
```

If you do not get a message such as `Discovered camera FTYB492321INQAX at 192.168.1.1`, then it means that this server is not able to communicate with any cameras, either they can't be found (wrong SSID, vlan, discovery network), the cameras are incompatible (eg: not A9) or there's a bug in the server.

Could you capture the network traffic when running this server (30 seconds should be enough) and upload the pcap file? You can use a program such as wireshark for this.

re: pairing: you should run `node dist/bin.cjs pair --ssid .. --password ...` as in the readme, but if the cameras are not discovered with `http_server`, it's unlikely that `pair` will do anything

danka621, 2024-07-07:
I also wondered about the SSID + password in the instruction.
Should I enter the SSID and password of my home network (router) or the SSID and password provided by the camera instruction to access its AP?

DavidVentura, 2024-07-07:
This SSID/password is your home's wifi, as the goal of the `pair` script is to configure the camera without using the app. You can also use the app to configure the cameras if you prefer, skipping the `pair` script

danka621, 2024-07-07:
Ok, I have not tried with the home wifi.. However, shouldnt the pair script discover my device anyway then? I run the pair script with ssid and password option and it get stuck in "searching for devices on 192.168.1.255".
(The PC has been assigned 192.168.1.100 when connecting to the camera AP, so I guess the script is searching in the right adress space..?)

DavidVentura, 2024-07-08:
To clarify:
- The camera is in Access Point mode
- You connect to the camera's access point with your laptop
- You ran the `pair` command, but did not manage to discover any camera

If this is correct, can you make a network capture using a program such as wireshark? it's important that you are connected to the camera's AP and you run the pair program for at least 30 seconds while recording

Leblayd, 2024-07-08:
Sorry for disappearing.
> what do you mean by nothing happens? do you get any output at all?
```
node dist/bin.cjs http_server
2024-07-08T14:50:40.402Z [info] Starting HTTP server on port 5000
2024-07-08T14:50:40.410Z [info] Searching for devices on 192.168.1.255
```
This is exactly what I get as well. Could have been more specific, but I didn't know if I was doing the right thing. In fact, I still don't.

Here is my network log. I stopped every process that tried anything, so it should be clean.
I did one try with the default config, and one with `--discovery_ip=192.168.169.255`, because of what `ip a` shows me: `inet 192.168.169.100/24 brd 192.168.169.255 scope global dynamic noprefixroute wlp0s20f0u13u2`
But again, I have no idea what I'm doing. Whatsoever.
[capture.zip](https://github.com/user-attachments/files/16131325/capture.zip)

danka621, 2024-07-11:
> To clarify:
>
> * The camera is in Access Point mode
> * You connect to the camera's access point with your laptop
> * You ran the `pair` command, but did not manage to discover any camera
>
> If this is [...]

Yes, that is correct. However I realized after checking the description of your project more in detail that it defines Ysxlite as the mobile application which is not the case for me. I also dont think the chip is TXW817, but another one (the battery covers most of the chip so cant see whats there). The SSID of the AP starts with DGOI-...

DavidVentura, 2024-07-11:
can you check the encoded branch? see 
https://github.com/DavidVentura/cam-reverse/issues/30#issuecomment-2220154935

danka621, 2024-07-11:
> can you check the encoded branch? see [#30 (comment)](https://github.com/DavidVentura/cam-reverse/issues/30#issuecomment-2220154935)

No difference to running the master branch.

I am running the code in wsl ubuntu on win 11 and connecting to the camera via windows. It says that I am given ip 192.168.1.100.
ip a in the wsl session says something but i guess there is a virtual adapter between wsl and windows. 
Anyway, I am able to ping 192.168.1.1 from ubuntu when connected to the ap, so cam-reverse should be able to see that network.