DavidVentura/cam-reverse #36: Two cameras not found

Bluscream, 2024-08-06:
I bought 2 of https://www.temu.com/goods.html?goods_id=601099566919307 and set both up via https://play.google.com/store/apps/details?id=com.macrovideo.v380pro and also i have a https://www.amazon.de/-/en/gp/product/B0BXNM48TD which uses https://play.google.com/store/apps/details?id=shix.vi.camera but none of the cameras are found in my wifi 
```bash
blu@xxx:/tmp $ node bin.cjs http_server --discovery_ip 192.168.2.255
2024-08-06T13:57:16.030Z [info] Starting HTTP server on port 5000
2024-08-06T13:57:16.058Z [info] Searching for devices on 192.168.2.255
```

DavidVentura, 2024-08-07:
Could you try the branch at #34 ? should be similar to PPPP, but I haven't tried it as I don't have any cameras with that protocol.

I can't access the pcap in that post, can you re-upload it somewhere?

Bluscream, 2024-08-07:
The pcap is included in the linked [github repo](https://github.com/Bluscream/cam-reverse-engineering/tree/main/Wireless%20Security%20Camera%20Three%20Antennas%20Wifi%20Surveillance%20-%20Temu)

Bluscream, 2024-08-27:
any news on this?

DavidVentura, 2024-08-29:
I'll look at the dumps today, but have just ordered a camera to test it out

Bluscream, 2024-08-29:
Very nice! Im looking forward to it <3 if you manage to get this working somehow i will help you get it into the homeassistant addon store

DavidVentura, 2024-08-29:
The [PCAPdroid_06_Aug_17_35_01.pcap](https://github.com/Bluscream/cam-reverse-engineering/blob/main/Wireless%20Security%20Camera%20Three%20Antennas%20Wifi%20Surveillance%20-%20Temu/PCAPdroid_06_Aug_17_35_01.pcap) in the repo looks to be a very different protocol than the PPPP proto (the only unencrypted messages have the strings `NVGROUPSEARCH` and NVDEVSEARCH``), maybe you can try something like [this](https://github.com/dunderhay/CCTV-v380-pro/blob/268a462debf5451fe939a37acc4cdb58d2dd96a4/scripts/discover-cam-on-network/findcam.py#L17)?

This camera (the v380 from temu) is the one I ordered, so we'll see, but I'm not expecting this project to work (maybe it'll start a new project..). 
Can I ask what led you to try the v380 with this project? Seems curious that it also happened in #41

The second camera ("Kavylany 1080P Dual Band Wireless WiFi Full Colour PTZ IP Camera Outdoor Night Vision Surveillance Dome Camera") is not shippable to Netherlands, so I can't try it (seems the most promising in this project if it works with PPPP)

Bluscream, 2024-08-29:
> The [...]

Interesting, i will take a look. Thanks for pointing this out. I assumed all these cheap china cams would use some form of PPPP
> This camera (the v380 from temu) is the one I ordered, so we'll see, but I'm not expecting this project to work (maybe it'll start a new project..).

I think a more modular approach that can handle more different protocols and convert them to something universal like Onvier would be much needed and a better longterm solution because it will be easier to maintain and extend
> Can I ask what led you to try the v380 with this project? Seems curious that it also happened in https://github.com/DavidVentura/cam-reverse/issues/41

Basically the same as in my first paragraph. There's nothing on the box that says "POWERED BY PROTOCOL XY" so i have to do some assumptions first and am glad if people can correct me
> The second camera ("Kavylany 1080P Dual Band Wireless WiFi Full Colour PTZ IP Camera Outdoor Night Vision Surveillance Dome Camera") is not shippable to Netherlands, so I can't try it (seems the most promising in this project if it works with PPPP)

That's also the one that works with 
- https://github.com/Bluscream/PPPP/
- https://github.com/Bluscream/PPPP/tree/python
- https://github.com/hyc/a9serv/tree/master

but these solutions both have downsides (PPPP_js is very unstable, PPPP_py i couldnt get working, a9serv only has a feed, no commands or anything else)

The only "nice" project i have found so far is https://github.com/dbuezas/ha-addons/tree/main/a9-cam which is able to be installed directly as homeassistant addon and can therefor run containerized by default. But it does not support **any** of the cameras i have.

Bluscream, 2024-08-29:
You certainly sent me on the right track

```log
PS C:\Users\blusc\AppData\Local\Temp\CCTV-v380-pro>  & 'c:\Users\blusc\AppData\Local\Programs\Python\Python313\python.exe' 'c:\Users\blusc\.vscode-insiders\extensions\ms-python.debugpy-2024.10.0-win32-x64\bundled\libs\debugpy\adapter/../..\debugpy\launcher' '51051' '--' 'C:\Users\blusc\AppData\Local\Temp\CCTV-v380-pro\scripts\discover-cam-on-network\findcam.py' 
[*] Looking for IP Camera on the local network...
[*] Looking for IP Camera on the local network...
[+] Camera found: b'NVDEVRESULT^200^4c:60:ba:21:98:08^192.168.2.95^255.255.255.0^192.168.2.1^5040^5050^5051^8800^7050^5040^84479835^0^1001'
```
Sadly i can't just connect to the cam with `rtsp://192.168.2.95:554` as the repo suggested:
```log

main error: connection failed: Connection refused by peer
satip error: Failed to connect to RTSP server 192.168.2.95:554
main error: connection failed: Connection refused by peer
access_realrtsp error: cannot connect to 192.168.2.95:554
```

DavidVentura, 2024-09-03:
I got the v380, but it does not seem to expose an access point, how did you connect to it?

Bluscream, 2024-09-03:
> I got the v380, but it does not seem to expose an access point, how did you connect to it?

didn't you got instructions and potentially a qr code for the app with it?
btw my discord is https://discord.com/users/467777925790564352