DavidVentura/cam-reverse #37: How all of this project works ?

ItsMeJacky, 2024-08-13:
Hello, sorry it will may be a dumb question, i'm very new to all this GitHub stuff, but... What are the prerequisites to install this project ?
I tried to clone the repository in a folder on a fresh debian OS and run "make build" but nothing happened. I think i'm missing something important but i can't find what ?

Can someone clearly explain me or provide me a link with some additionnal explanation ?

Thank you and sorry for this dumb question.

DavidVentura, 2024-08-13:
You can download the built files from the releases page and run it as indicated in the readme (search http_server)

ItsMeJacky, 2024-08-13:
I started the server, i missed the installation of "npm", server running now.
Now i'm struggling to scan wifi network to find my A9 cam... but this is not a problem of your project.
I will back if i nedd more help, thank for your reactivity !

closed 2024-08-29 by DavidVentura