DavidVentura/cam-reverse #40: stream usage with go2rtc or VLC

b2un0, 2024-08-21:
hey there,

im using for all my different cameras [go2rtc](https://github.com/AlexxIT/go2rtc) so i can multiplex the streams to eg. [frigate](https://frigate.video), as native apple homekit device and in Home Assistant

but, your implementation only outputs a endless jpg stream.

i have absolutely no idea how to use this as jpg or ffmpeg input in `go2rtc` or view the stream with VLC

a native mjpg stream which audio included would be nice.

DavidVentura, 2024-08-21:
> your implementation only outputs a endless jpg stream.
Isnt this what mjpeg is? ffmpeg can play it

do you get any specific error?

>  native mjpg stream which audio included would be nice.
mjpeg is not usually put in any container, so audio data can't be part of the stream

unless you mean packing the streams in something like matroska?

b2un0, 2024-09-05:
you're right. my mistake.

the mjpg stream seems to be correct.

but i cant get it work with `go2rtc`.

but I'm giving up for now and switching back to an ESP32 cam

closed 2024-09-05 by b2un0