DavidVentura/cam-reverse #41: firmware dump of v380 blub e27 IoT WiFI smart camera

parsapoorsh, 2024-08-21:
hi, i have a firmware dump of a v380 blub e27 IoT WiFI smart camera.
[image]
this camera uses a `T25S80` flash chip, with a `XF16 P9141EA6AD1` SoC.
i thought you or someone else might need it
name: `v380-blub-IoT-camera-firmware.bin`
sha256: `28c1bd4fa632d7ba30148c0c51acd15a7844b90bb06db83e254c8be7c434cdb8`
download link: [Google Drive](https://drive.google.com/file/d/1iSVKfRViHmEZS_RndtezoDl4VsZwCEQU/view?usp=sharing)

parsapoorsh, 2024-08-30:
> I'm not really sure v380 runs a similar protocol to the X5

i was able to use this project to get video stream from the camera. audio didn't worked, and PTZ control was missing.
but that doesn't matter because the camera firmware itself is buggy and crashes, even when using the app

divadiow, 2025-04-14:
newer SDK xradio_skylark_sdk-1.2.2 - https://github.com/divadiow/xr872_sdk