DavidVentura/cam-reverse #44: cannot pair the camera to my wifi

ICrashWindows12, 2024-09-21:
Model: A9
Manufacturer: Telurahl (it states on packaging ) (I also don't know if I got the manufacturer name correct)
A9 AP starts with: Nax (short for naxclow)
My laptop OS: ChromeOS with Linux Container Bullseye/Crostini
CPU arch: ARM64
I cannot pair for some reason. The IP address starts at 192.168.169.1 but when I pair it states it looks for devices at 192.168.1.255. I am connected to the AP of the camera. is there any way to change. Also is this compatible with chromeOS running Linux debian vm (the command line) 

2024-09-21T11:06:16.808Z [error] Node version 12 is not supported, may malfunction
2024-09-21T11:06:16.811Z [info] Will configure any devices found to join Optus_XXXXXX
2024-09-21T11:06:16.848Z [info] Searching for devices on 192.168.1.255

ICrashWindows12, 2024-09-21:
does changing the bin.cjs file even do anything. I edited it with `sudo nano bin.cjs` then I found the discovery ips section and changed from 192.168.1.255 to 192.168.169.255, then I tried 192.168.169.1 and still nothing

ICrashWindows12, 2024-09-21:
I executed the command like this `node bin.cjs pair --ssid Optus_XXXXX --password <Password>` obviously with the password after I connected to the devices AP.

ICrashWindows12, 2024-09-21:
I'm going to see if I can connect the X5 mini camera

ICrashWindows12, 2024-09-21:
also NAX

ICrashWindows12, 2024-09-21:
nope. maybe the restrictions on chromeOS's Linux container does not allow it to discover devices. also the one by int is just so confusing

DavidVentura, 2024-09-21:
can you make a network capture while connected to the camera's AP?

ICrashWindows12, 2024-09-21:
how do I make a network capture

ICrashWindows12, 2024-09-21:
> can you make a network capture while connected to the camera's AP?

while connected to the cameras AP?

ICrashWindows12, 2024-09-21:
> can you make a network capture while connected to the camera's AP?

How do I make a network capture. I just found out about network capturing and I'm new to it

ICrashWindows12, 2024-09-21:
> can you make a network capture while connected to the camera's AP?

do I run the script along with wireshark

ICrashWindows12, 2024-09-21:
> can you make a network capture while connected to the camera's AP?

[image]

ICrashWindows12, 2024-09-21:
i did it on windows because I couldn't figure it on chromeOS

ICrashWindows12, 2024-09-23:
> can you make a network capture while connected to the camera's AP?

I did already. was it supposed to be after I ran the script

ICrashWindows12, 2024-09-23:
> can you make a network capture while connected to the camera's AP?

PLEASE RESPOND. I may have found something
[image]
[image]
. I don't know if it is useful info.

rundekugel, 2024-10-20:
1st: David: thanks for your work!

similar problem here.
the param `--discovery_ip 192.168.1.1`  is not working for me with command 'pair'
changed `discovery_ips` in bin.cjs
camera is found, but :

```
Logging in to camera BATE637140UAGEB
2024-10-20T18:20:15.045Z [info] Scanning for Wifi networks on BATE637140UAGEB -- this may time out
2024-10-20T18:20:22.690Z [info] Camera BATE637140UAGEB at 192.168.1.1 already discovered, ignoring
2024-10-20T18:20:23.989Z [info] Camera BATE637140UAGEB at 192.168.1.1 already discovered, ignoring
2024-10-20T18:20:25.399Z [warning] Camera BATE637140UAGEB timed out
2024-10-20T18:20:25.399Z [info] Disconnected from camera BATE637140UAGEB at 192.168.1.1
2024-10-20T18:20:25.400Z [info] Camera BATE637140UAGEB disconnected
2024-10-20T18:20:26.991Z [info] Discovered camera BATE637140UAGEB at 192.168.1.1
2024-10-20T18:20:27.045Z [info] Logging in to camera BATE637140UAGEB
```
Then I reconnected wifi, started bin.cjs again... it worked.

kohend, 2024-10-20:
Changed in bin.cjs and it works for me as well, thanks @rundekugel