DavidVentura/cam-reverse #46: A9/X5 variants

divadiow, 2024-10-03:
I've 5 of these. 4 A9s with Allwinner/Xradiotech XF16 and Taixin TXW817-810 and an X5 with TXW817-810.

The XF16 has an external 1mb SOIC8 flash chip, so here's the dump from that
[Flashrom_C74014.bin.zip](https://github.com/user-attachments/files/17249809/Flashrom_C74014.bin.zip)

Some musings, including boot logs in https://www.elektroda.com/rtvforum/topic4074636.html and https://www.elektroda.com/rtvforum/topic4033757.html#21035231 where you'll also see I unsuccessfully tried to get to the internal flash on one of the cams using CH341A, After reading this though I know why that's not an option https://habr.com/ru/articles/810291/

Maybe I'll try the ST bluepill idea

I've yet to try cam-reverse. Is it expected to work for all MCU variants?

I was going to hook them all up to certmitm/mitmrouter and wireshark to see if anything new could be seen. let me know if any of this is of interest and if you want anything specific.

[bootlog.txt](https://github.com/user-attachments/files/17249877/bootlog.txt) - from green PCB TXW817

DavidVentura, 2024-10-03:
Not all are expected to work, but it seems unrelated to the MCU. Some firmware versions encrypt the payload with a sinmple scheme, but that's not implemented, as neither of my 5 cameras have that firmware version.

jxoffline, 2025-01-24:
Any chances to support BEKEN IC (.i.e BK7252) as well?

divadiow, 2025-05-25:
First dump of a TXW817 firmware https://www.elektroda.com/rtvforum/topic4033757.html#21558806

divadiow, 2025-05-27:
Taixin TXW81X flash backup procedure https://www.elektroda.com/rtvforum/topic4123724.html

jmgk77, 2026-11-05:
For the X5/YsxLite variant (serials starting with `BATE`), the payload transform is implemented and documented: invert bit 0 of each byte, then rotate the buffer by 4 bytes (`XqBytesEnc`/`XqBytesDec`). The login is a fixed 180-byte Drw copied from the official app, and the ticket comes from `ConnectUserAck` at offset 24 after decrypting the 8-byte payload.

I ported this to ESP32-C3/S3/CAM/DevKit and kept a Linux reference: https://github.com/jmgk77/x5-bridge

I'm the author. I only tested `BATE558596HDLZB` and `BATE583547ANCGG`, so other X5 firmware variants may differ.