DavidVentura/cam-reverse #48: Add error handling

adlerweb, 2024-12-12:
When the script encounters a stalled or unpaired camera, it will crash. If you use multiple cameras, you won't be able to use any of them until you removed the offending camera. Instead of crashing, the script should use some sort of try-block to catch these errors and just ignore the camera (add to blacklisted_ips?) instead of crashing.

Example (one camera stalled):

```
$ node dist/bin.cjs http_server --discovery_ip 192.168.99.255
2024-12-12T00:22:19.820Z [info] Starting HTTP server on port 5000
2024-12-12T00:22:19.827Z [info] Searching for devices on 192.168.99.255
2024-12-12T00:22:22.841Z [info] Discovered camera TBAT… at 192.168.99.234
2024-12-12T00:22:22.887Z [info] Discovered camera TBAT… at 192.168.99.69
/home/adlerweb/dev/cam-reverse/dist/bin.cjs:24517
  return new DataView(this.buffer, offset + this.byteOffset);
         ^

RangeError: Start offset 24 is outside the bounds of the buffer
    at new DataView (<anonymous>)
    at DataView.add (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:24517:10)
    at createResponseForControlCommand (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:24805:33)
    at Object.handle_Drw [as Drw] (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:25004:15)
    at handleIncoming (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:25024:16)
    at Socket.<anonymous> (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:25037:38)
    at Socket.emit (node:events:517:28)
    at UDP.onMessage [as onmessage] (node:dgram:942:8)

Node.js v18.20.5
```

DavidVentura, 2024-12-12:
can you run in debug mode? seems like the control command received is corrupt/too short

adlerweb, 2024-12-19:
Sure. In this case, it is just a camera (…ADEF) not yet paired to cam-reverse (=using a different password/key/whatever) in the same network. My suggestion is to gracefully ignore devices replying bogous packages. Currently cam-reverse crashes, as such I can't access the working cameras (…AEET) as long as a non-working one is in the same discovery range.

```
2024-12-19T02:27:13.989Z [info] Starting HTTP server on port 5000
2024-12-19T02:27:13.997Z [info] Searching for devices on 192.168.99.255
2024-12-19T02:27:14.004Z [debug] Received a PunchPkt message
2024-12-19T02:27:14.005Z [info] Discovered camera TBAT…ADEF at 192.168.99.234
2024-12-19T02:27:14.012Z [debug] Sending Drw Packet with id 0
2024-12-19T02:27:14.025Z [debug] Removing 0 from pending
2024-12-19T02:27:14.071Z [debug] Received a PunchPkt message
2024-12-19T02:27:14.072Z [info] Discovered camera TBAT…AEET at 192.168.99.69
/home/adlerweb/dev/cam-reverse/dist/bin.cjs:24517
  return new DataView(this.buffer, offset + this.byteOffset);
         ^

RangeError: Start offset 24 is outside the bounds of the buffer
    at new DataView (<anonymous>)
    at DataView.add (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:24517:10)
    at createResponseForControlCommand (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:24805:33)
    at Object.handle_Drw [as Drw] (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:25004:15)
    at handleIncoming (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:25024:16)
    at Socket.<anonymous> (/home/adlerweb/dev/cam-reverse/dist/bin.cjs:25037:38)
    at Socket.emit (node:events:517:28)
    at UDP.onMessage [as onmessage] (node:dgram:942:8)

Node.js v18.20.5
```