DavidVentura/cam-reverse #50: Requested door

Aieser, 2025-01-09:
I have a problem; I think the requested door is not the correct one. How can we check it?

Model with TXW817 chip, HDWIFICAMPRO app.

[image]

[image]

[image]

DavidVentura, 2025-01-10:
Did you connect to the camera's AP? What is your  lan IP? and the camera's?

Aieser, 2025-01-10:
Did you connect to the camera's AP? 
_Yes, my conputer is conected in DGOF-470816-PZLPK, camera AP SSID_

 What is your lan IP?
 _192.168.1.100_

and the camera's?
_192.168.1.1_