DavidVentura/cam-reverse #51: Encoded JSON-based protocol for PPPP cameras

devbis, 2025-01-20:
I have a DG** camera  (DGOK in my case) which uses vi365 app as the stock application. 
I assumed I could use it with this app and failed. 
The result was the same as in https://github.com/DavidVentura/cam-reverse/issues/30, the app sends unencrypted discovery packets... and that's all.
So I took your `encoded` branch as the start point. Some initial fixes I implemented in two commits here (https://github.com/devbis/cam-reverse/commits/encoded-discovery/). But more work to be done: requesting video/audio, handling frames, etc.

I would like to support both old and new JSON protocols.

1. This app doesn't send encrypted LanSearch packets. I added sending both encrypted and unencrypted. Camera started to send Punch packets.
2. Camera on login asked for closing the connection. I changed 
- instead of P2pRdy packet I'm sending PunchPkt back
- login is not needed for my cam, it sends ClosePkt for legacy login command. (maybe for other legacy commands too) 

Then, it uses json payload in DRW packets to ask for streams. JSON proto is implemented in https://github.com/Bluscream/PPPP/blob/master/pppp.js . It also has downsides, like my video frame marker is '55aa15a80303' instead of '55aa15a80300' in the file. 

I implemented a python library that works with my camera https://github.com/devbis/aiopppp/ based on your structure and Bluscream's protocol implementation. But it would be nice to add this functionality here.

DavidVentura, 2025-01-20:
Can you send me a link where you bought the camera?

devbis, 2025-01-20:
It was a local store, but it looks exactly like https://www.aliexpress.com/item/1005007743948353.html

magicus, 2025-01-20:
Hi, magicus here, author of https://github.com/magicus/PPPP upon which https://github.com/Bluscream/PPPP is based. I kind of gave up on the project, but I'd still like to see something usable come out of the effort that went into it. Let me know if I can be of any assistance here.

magicus, 2025-01-20:
@devbis Sounds like you have the same plans as I did.

Yes, I also got problems with the connection being dropped. I started debugging rewriting the networking in the javascript program I based my efforts on, but I thought it seemed like wasted time when my ultimate goal was to have this written in Python instead. So I managed to create a cookiecutter repo at https://github.com/magicus/pypppp, but then I got stuck on network handling with Python, as I had to little experience for that. :-/

I think some kind of heartbeat would solve the problem; or in the worst case, a "watchdog" which notices when the connection is lost and immediately reconnects.

magicus, 2025-03-10:
@devbis Wow, that is awesome! I can't wait to try it out!

Bluscream, 2025-03-14:
Sadly this doesnt work with my PPPP cam :(

shlomicthailand, 2025-03-20:
> you check the repos from devbis post

Yes i did , pretty quick too :-) 
Unfortunately it did not discover the camera - i'm not sure if this is because it already paired with the project from this repo 
(cam-reverse)

devbis, 2025-03-24:
@shlomicthailand  Hi! You may join to test binary protocol implementation based on this project. The discussion and steps are here https://github.com/devbis/aiopppp/issues/3
I just need to collect logs in both cases if it works and if doesn't.
Later, it will be added to the HA component.

YeonV, 2025-03-26:
> [@magicus](https://github.com/magicus) I created a library to communicate with my camera https://pypi.org/project/aiopppp/0.1.0/ and wrote a basic custom component https://github.com/devbis/pppp_camera/ to add cameras to Home Assistant. It might be not of the best quality (it is my first component) but it works.

hi bro. after having the JSON error with your pppp, i quickly created an home assistant addon, which runs @DavidVentura's cam-reverse inside home assistant. Its working as a proof-of-concept. It can discover the cams and serve the streams.
i just placed this log, so users can create camera entities:

[image]

so adding this as a MJPEG IP Camera, is working great.
Then, as you can see i started to auto-create the camera entity, and something also happened:

[image]

but sadly that one is not working. (the manually added ones are working)

addon-repo:
https://github.com/YeonV/home-assistant-addons/

addon-source:
https://github.com/YeonV/home-assistant-addons/tree/master/nodejs-server

cam-reverse slightly extended:
https://github.com/YeonV/cam-reverse

devbis, 2025-03-26:
https://github.com/devbis/pppp_camera/releases/tag/1.1.0

I added initial support for binary protocol in the custom component. 
You need to provide credentials, sometimes it is admin:admin, sometimes admin:6666. Or maybe different.