DavidVentura/cam-reverse #55: Not able to add A9 camera

xacb33, 2025-03-05:
Hello, I know this project has not been updated in a while, but I hope someone can help me with this please :)

I followed all the steps to connect to my camera, but I wasn't able to.

Connected to the camera's AP, ran the script to add the camera... and nothing.

(Also, I tried disabling Windows firewall, but nothing)

[image]

Thank you very much for this awesome project!

DavidVentura, 2025-03-05:
can you try this branch ? 
https://github.com/DavidVentura/cam-reverse/pull/52

xacb33, 2025-03-07:
> can you try this branch ? [#52](https://github.com/DavidVentura/cam-reverse/pull/52)

I wasn't able to add it, but I was able to connect by first pairing it with YsxLite, looking the A9's IP and adding it in the custom config.

For me it's ok like this, but I'm available if you want to test with my camera (Newer firmware maybe?)

Again, thank you very much ffor this project! Once connected it works perfectly (The bad image quality is obviously because the camera was 1.99€)

DavidVentura, 2025-03-07:
Right, then in this case, probably the discovery IP used by default was incorrect.