DavidVentura/cam-reverse #57: Infrared mode support

shlomicthailand, 2025-03-21:
Hi 
I have this project running with my [camera](https://fr.aliexpress.com/item/1005008153468173.html?spm=a2g0o.order_list.order_list_main.5.1ab01802FGe9Pi&gatewayAdapt=glo2fra) (not classic A9 but i guess similar) 

i did notice the original crappy app had night mode which turned on the infra-red , what would it take to add it ?
if you can give me a direction maybe i can add it myself , because looking at the code looked kind of cryptic without knowing the protocol 

thanks
Shlomi

DavidVentura, 2025-03-21:
Can you send a photo of the camera? I had infrared support in this repo, but all the camera does is turn on the grayscale filter.

My cameras do not have IR LEDs on them, even if the app had a "night mode".

shlomicthailand, 2025-03-21:
> Can you send a photo of the camera? I had infrared support in this repo, but all the camera does is turn on the grayscale filter.
> 
> My cameras do not have IR LEDs on them, even if the app had a "night mode".

You have it in the link to Ali , but here are couple more 

[image]

[image]

shlomicthailand, 2025-03-21:
@DavidVentura - quick question cause i'm not sure i got this about this camera.
are we saying it does not support RTSP in any way ?
only their binary proprietary protocol ?

DavidVentura, 2025-03-21:
That's right. You could probably write an RTSP adapter for this server if you want this protocol.

shlomicthailand, 2025-03-21:
I'm looking to show these in HA (or any app that support them which is not YsxLite) , i bought a bunch thinking to conceal them in all sort of thing in the house .not sure whats the best course of action. 
They were not so cheap as the A9 - i paid 8$ for each .

DavidVentura, 2025-03-21:
can you use this? https://www.home-assistant.io/integrations/mjpeg/ this project just exposes an mjpeg stream

shlomicthailand, 2025-03-21:
Yes i wanted to , but the stream (even in your html page locally) is very laggy , and sometimes stuck (not sure if its the cam or the node server)

shlomicthailand, 2025-03-21:
> [@DavidVentura](https://github.com/DavidVentura) - quick question cause i'm not sure i got this about this camera. are we saying it does not support RTSP in any way ? only their binary proprietary protocol ?

@DavidVentura - i understand its not supporting RTSP but do you know why this command returns possible RTSP commands
```
dev curl -v -X DESCRIBE "rtsp://admin:admin@192.168.1.233:7070/"      
*   Trying 192.168.1.233:7070...
* Connected to 192.168.1.233 (192.168.1.233) port 7070
* Server auth using Basic with user 'admin'
> OPTIONS * RTSP/1.0
> CSeq: 1
> User-Agent: curl/8.7.1
> Authorization: Basic YWRtaW46YWRtaW4=
> 
* Request completely sent off
< RTSP/1.0 200 OK
< CSeq: 1
< Public: DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE
< 
* Connection #0 to host 192.168.1.233 left intact
```

DavidVentura, 2025-03-21:
Maybe your camera/firmware version supports RTSP! Try it out with ffmpeg orother players. My cameras don't support it.

shlomicthailand, 2025-03-21:
@DavidVentura - maybe you can give a direction, this is what i got from ffplay 
```
ffplay -analyzeduration 2147483647 -probesize 2147483647 rtsp://192.168.1.233:7070/webcam
ffplay version 7.1.1 Copyright (c) 2003-2025 the FFmpeg developers
  built with Apple clang version 16.0.0 (clang-1600.0.26.6)
  configuration: --prefix=/opt/homebrew/Cellar/ffmpeg/7.1.1_1 --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags='-Wl,-ld_classic' --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libaribb24 --enable-libbluray --enable-libdav1d --enable-libharfbuzz --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox --enable-audiotoolbox --enable-neon
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
**[rtsp @ 0x1348107d0] Could not find codec parameters for stream 0 (Video: mjpeg, none(bt470bg/unknown/unknown)): unspecified size
Consider increasing the value for the 'analyzeduration' (2147483647) and 'probesize' (2147483647) options
Input #0, rtsp, from 'rtsp://192.168.1.233:7070/webcam':**
  Metadata:
    title           : Test
  Duration: N/A, bitrate: 128 kb/s
  Stream #0:0: Video: mjpeg, none(bt470bg/unknown/unknown), 90k tbr, 90k tbn
  Stream #0:1: Audio: pcm_s16be, 8000 Hz, mono, s16, 128 kb/s
    nan A-V:    nan fd=   0 aq=    0KB vq=    0KB sq=    0B 
```

if i remove the path `webcam` i get 404 - so i guess it does have something there

DavidVentura, 2025-03-21:
it's finding video and audio streams, but can't identify them. You will need to experimeny with format specifiers, probably it's mjpeg, 20/30fps and 640x480, but no way to tell

shlomicthailand, 2025-03-22:
Thanks @DavidVentura 
last question hopefully - when  a camera is connected both to app i get this error trying to start the server.
any ideas ?
```
2025-03-22T15:03:56.211Z [info] Discovered camera TBAT557046EYJOT at 192.168.1.233
/Users/shlomic/dev/github/cam-reverse/dist/bin.cjs:24517
  return new DataView(this.buffer, offset + this.byteOffset);
         ^

RangeError: Start offset 24 is outside the bounds of the buffer
    at new DataView (<anonymous>)
    at DataView.add (/Users/shlomic/dev/github/cam-reverse/dist/bin.cjs:24517:10)
    at createResponseForControlCommand (/Users/shlomic/dev/github/cam-reverse/dist/bin.cjs:24805:33)
    at Object.handle_Drw [as Drw] (/Users/shlomic/dev/github/cam-reverse/dist/bin.cjs:25004:15)
    at handleIncoming (/Users/shlomic/dev/github/cam-reverse/dist/bin.cjs:25024:16)
    at Socket.<anonymous> (/Users/shlomic/dev/github/cam-reverse/dist/bin.cjs:25037:38)
    at Socket.emit (node:events:507:28)
    at UDP.onMessage [as onmessage] (node:dgram:990:8)

```

YeonV, 2025-03-24:
@DavidVentura EYYY, WONDERFUL!!!!

thank you sooo much for sharing this project OMG!!!

@shlomicthailand 
added 4 of these micro-beasts to HomeAssistant via "MJPEG IP Camera" Integration

[image]

[image]

shlomicthailand, 2025-03-24:
@YeonV - yes me too, but i wanted to avoid the node server running all the time. (externally to HA)

YeonV, 2025-03-24:
> This project aim to do it in python as HA addon [devbis/aiopppp#3](https://github.com/devbis/aiopppp/issues/3)

that didnt work for me because of some JSON error

YeonV, 2025-03-24:
@shlomicthailand okaaaay. have this now up and running within homeassistant

YeonV, 2025-04-05:
@shlomicthailand 

[image]