DavidVentura/cam-reverse #60: PTZ support

skinkie, 2025-05-09:
I did not find any issues regarding pan and tilt on some camera's.

My guess is that CMD_PASSTHROUGH_STRING_PUT in the dissector likely shows the what happens:

```
00-0b-51-48-01-11-fe-fe-01-01-01-01-01-01-01-01-01-01-01-00-01-01-01-01-19-01-01-01
00-0b-50-48-11-01-fe-fe-01-01-01-01-01-01-01-01-01-01-01-00-01-01-01-01-19-01-01-01
00-0b-51-48-01-11-fe-fe-01-01-01-01-01-01-01-01-01-01-01-0d-01-01-01-01-19-01-01-01
00-0b-50-48-11-01-fe-fe-01-01-01-01-01-01-01-01-01-01-01-0d-01-01-01-01-19-01-01-01
...
00-0b-50-48-11-01-fe-fe-01-01-01-01-01-01-01-01-01-01-01-03-01-01-01-01-19-01-01-01
```

This does not occur when no PTZ action has been touched.

DavidVentura, 2025-05-09:
Can you link what camera you got with PTZ support? This repo only supports the cameras I have access to

skinkie, 2025-05-10:
https://nl.aliexpress.com/item/1005007486153427.html

DevLn, 2025-05-17:
You can check my python implementation [here](https://github.com/DevLn/aiopppp/commit/f52397491a6a39dfa43552c3ad362f311970f2cd)

skinkie, 2025-05-17:
@DevLn at your implementation I am missing the audio. While it does not work here either (in the app it does), it would be nice to get resolved.

DevLn, 2025-05-17:
I've just added implementation for PTZ in aiopppp lib. I may implement some other features at some point if have free time. For now I'm not planning to do so.