DavidVentura/cam-reverse #61: The camera is discovered in the local LAN, but fails parsing the received packet

cristeab, 2025-07-30:
Hi

I have a Xiaomi X5 Mini Camera already configured to connect to my LAN. The camera IP address is known and put in a config file.
With the code from master branch I am starting the web server with: 

`node dist/bin.cjs http_server --config_file ../TBAT944135XOFUO.yml`

but parsing the received packet from the camera fails:

2025-07-30T19:21:03.517Z [info] Starting HTTP server on port 5000
2025-07-30T19:21:03.541Z [info] Searching for devices on 192.168.77.39
2025-07-30T19:21:03.547Z [debug] Received a PunchPkt message
2025-07-30T19:21:03.548Z [info] Discovered camera TBAT944135XOFUO at 192.168.77.39
2025-07-30T19:21:03.555Z [debug] Sending Drw Packet with id 0
2025-07-30T19:21:03.565Z [debug] Removing 0 from pending
/home/bogdan/projects/cam-reverse/dist/bin.cjs:24517
  return new DataView(this.buffer, offset + this.byteOffset);
         ^

RangeError: Start offset 24 is outside the bounds of the buffer
    at new DataView (<anonymous>)
    at DataView.add (/home/bogdan/projects/cam-reverse/dist/bin.cjs:24517:10)
    at createResponseForControlCommand (/home/bogdan/projects/cam-reverse/dist/bin.cjs:24805:33)
    at Object.handle_Drw [as Drw] (/home/bogdan/projects/cam-reverse/dist/bin.cjs:25004:15)
    at handleIncoming (/home/bogdan/projects/cam-reverse/dist/bin.cjs:25024:16)
    at Socket.<anonymous> (/home/bogdan/projects/cam-reverse/dist/bin.cjs:25037:38)
    at Socket.emit (node:events:517:28)
    at UDP.onMessage [as onmessage] (node:dgram:942:8)

Node.js v18.19.0

tomgat6, 2025-08-26:
I had the same issue and debugged it: It happened because my camera didn't use "admin"/"admin" for user/password, which is hardcoded when http_server mode is logging in to the camera. (handler.ts line 29)
Try resetting the camera and pairing it via this cam-reverse utility as described in the readme. (https://github.com/DavidVentura/cam-reverse#pairing-a-new-camera)

jmgk77, 2026-11-05:
Different X5 variant, but the offset may be related. In my X5/YsxLite capture the login is a fixed 180-byte Drw taken verbatim from the official app, and the ticket is read from `ConnectUserAck` at offset 24 after decrypting the 8-byte payload. `payload_len` at `[12..13]` is little-endian there. If the received packet is shorter than 28 bytes, reading offset 24 is out of bounds, so guard it with `n >= 28`.

My working ESP32/Linux parser and the protocol notes are at https://github.com/jmgk77/x5-bridge (I'm the author). I can't confirm it is the same firmware as your X5 Mini, so treat the offset detail as a hint.