DavidVentura/cam-reverse #63: PTZ Outdoor cameras

ElYaiko, 2025-08-15:
It would be possible to support this functions from the YSXLite app for PTZ Cameras ? 
- Vertical and horizontal rotation
- Toggle white and IR light

This are not very important but it'd be nice too:
- Stream mic/audio
- Fix audio (I can't hear nothing with this camera)
- Changing user password
- Reboot

I have this "amzwn" PTZ Camera (https://nl.aliexpress.com/item/1005007486153427.html)
Thanks a lot!

DavidVentura, 2025-08-15:
Hi
Yes, it would be possible.
I got a PTZ camera from aliexpress that I still need to test out.

ElYaiko, 2025-08-19:
> Hi Yes, it would be possible. I got a PTZ camera from aliexpress that I still need to test out.

Nice!, please release the PTZ rotation first 🙏