DavidVentura/cam-reverse #65: Reverse Engineering tips on A9

d1nuc0m, 2025-09-10:
Hi, I'm trying to experiment with an A9,  see you were able to extract some logs using UART, which pads did you use? I tried probing but wasn't able to locate the right ones and they aren't marked

Also, were you able to extract the fw or only reversed the companion app?

divadiow, 2025-09-10:
but which A9 do you have?

d1nuc0m, 2025-09-10:
@divadiow it looks quite like the one in pics folder

[image]
[image]

The main chip is a TXW817 (TXW817-810 CWCN4A G2523)

divadiow, 2025-09-10:
interesting. I've not seen this blue PCB variation before.

I reckon TX is most likely to be the DM pad = HSDM/PC7

921600 baud

see https://www.elektroda.com/rtvforum/viewtopic.php?p=21560895#21560895